balena-os/meta-balena #4: config.json has these ugly `\n` newline chars

shaunmulligan, 2015-12-09:
Currently the config.json has these ugly `\n` newline chars that are painful for the users to work with when changing network settings:
here is an example:
```
"files": {
    "network/settings": "[global]\nOfflineMode=false\n\n[WiFi]\nEnable=true\nTethering=false\n\n[Wired]\nEnable=true\nTethering=false\n\n[Bluetooth]\nEnable=true\nTethering=false",
    "network/network.config": "[service_home_ethernet]\nType = ethernet\nNameservers = 8.8.8.8,8.8.4.4\n\n[service_home_wifi]\nType = wifi\nName = My_Wifi_Ssid\nPassphrase = my super secret wifi passphrase\nNameservers = 8.8.8.8,8.8.4.4"
  }
```

emirotin, 2016-06-07:
We could switch to YAML that has multiline strings

imrehg, 2018-01-23:
@shaunmulligan this is not relevant for resinos 2.x anymore, right?

shaunmulligan, 2018-01-23:
@imrehg that is correct. we can close

closed 2018-01-23 by imrehg