balena-os/meta-balena #11: resin-device-register is present on non-flasher images

floion, 2016-02-15:
We should only have resin-device-register on flasher images where supervisor is not running. On non-flasher images, where supervisor is running we do not register using this script, instead registration is done by the supervisor.

resinhup has resin-device-progress as a dependency which in turn needs resin-device-register, so this is the dependency chain which leads to having resin-device-register in non-flasher images.

resin-device-register will fail on these non-flasher images as registration is already done by the supervisor.

floion, 2016-02-15:
@telphan @agherzan @petrosagg

agherzan, 2016-02-15:
Well we actually need resin-device-register now on normal images too. And that is for resinhup tool which needs to send tokens / states to API.

floion, 2016-02-15:
I believe you meant resin-device-progress is needed for resinhup to send states to API, and not resin-revice-register. Right?

resin-device-register's work is done by the supervisor

@pcarranzav confirm?

agherzan, 2016-02-15:
Right. I confirm. I see now the chain. Maybe we should just break this dependency and install them separately as needed.

closed 2016-02-29 by agherzan