balena-os/meta-balena #12: Splash

agherzan, 2016-02-15:
# Steps to reproduce issue 
Connect to the HDMI connected to a monitor
Boot the device

# Expected result 
Resin logo splash screen should be visible

# Actual result 
Logo is not visible

# Other information 
Reported by: Florin Sarbu 

Test configuration: VIA VAB820 

Test case number: TC07 

Test run: [https://resinio.testlodge.com/projects/16238/runs/192879?tab=2&run_section_id=168785&executed_case_id=9333902#executed_case_9333902](https://resinio.testlodge.com/projects/16238/runs/192879?tab=2&run_section_id=168785&executed_case_id=9333902#executed_case_9333902)

floion, 2016-02-15:
plymouth should only be visible for production images

On the tested image (which is a staging image) the image rendered by plymouth gets garbled with kernel boot log messages.

agherzan, 2016-02-15:
@floion could you please move this to https://github.com/resin-os/resin-via-arm ?Thanks. When done, please close this one.

floion, 2016-02-15:
I have opened it as a result of a via test run, but it should really belong here

agherzan, 2016-02-15:
Isn't this a via specific bug? @floion

floion, 2016-02-15:
plymouth is started on all the staging images

telphan, 2016-06-07:
Please note that the testlodge tests need adjustments after this issue is closed.

agherzan, 2016-06-07:
Before*

agherzan, 2017-07-18:
This board was discontinued so I'm closing this bug. https://github.com/resin-os/resin-via-arm

closed 2017-07-18 by agherzan