balena-os/meta-balena #19: Serial console removal in production should be done with systemd masking

agherzan, 2016-02-19:
For easy undo-ing of this feature, it would be better to use systemd mask for these services.

telphan, 2016-02-19:
This is something I wanted to do for a long time now. I will take care of it.

agherzan, 2016-02-29:
@telphan , @floion agreed to fix this.

floion, 2016-03-01:
@agherzan , @petrosagg , @telphan I believe we should better off remove the kernel cmdline "console" argument for production altogether instead of masking. Masking this serial getty service is not possible, as systemd instantiates this service based on the console for each particular device

agherzan, 2016-03-02:
In that case we need a unified way of manipulating command line arguments. As well something that would be able to switch to a debug mode.

agherzan, 2016-03-07:
@floion are you sure systemd starts getty on all console arguments? I don't find any documentation about this and don't seem to be able to replicate on device.

telphan, 2016-05-09:
@agherzan I am pretty sure it does. You can check the current rpi-basic-image in meta-rpi, that misses the console=tty1 argument and no getty gets spawned before is is added to the kernel command line.

agherzan, 2016-05-09:
Here is the information that I was looking for @floion http://0pointer.net/blog/projects/serial-console.html . We need to sync on this @telphan @floion . I agreed that on production there should be no console kernel argument to avoid having getty spawned on those. But this is somehow a different issue than the one where systemd-serialgetty.bb enables the ones in SERIAL_CONSOLE which we currently handle my removing the files.

agherzan, 2017-07-18:
This was fixed in the meanwhile by removing the getty systemd generator.

closed 2017-07-18 by agherzan