balena-os/meta-balena #31: connman uses only 3 candidates for nameservers which are chosen non-deterministic

agherzan, 2016-03-02:
If DHCP gives 2 IPv4, say 75.75.75.75 and 76.76.76.76 and 2 IPv6 ones then we have  total of 6 DNS nameservers. Only 3 of them will get serialised to /etc/resolv.conf and in no particular order. That means that a device that got the network configuration once and connected successfully isn't guaranteed to do the same in the event of a reboot or a carrier loss. 

Also, we specify 8.8.8.8 and 8.8.4.4 both in /etc/connman/main.conf as FallbackNameservers and in /var/lib/connman/network.config in Nameservers but that doesn't seem to imply anything in terms of priority.

agherzan, 2016-04-08:
@floion this was implemented by dnsmasq. Please confirm and close this issue. It would be good to mention the issues in the future PRs so we can trace and link them.

floion, 2016-04-26:
This can be closed

closed 2016-04-26 by agherzan