balena-os/meta-balena #69: Support for additional wifi chipsets

shaunmulligan, 2016-05-12:
Namely looking for support for:
* rt2800usb
* RTL8812AU

mathewpeterson, 2016-12-22:
It would be nice to support [HubPiWi Blue](http://referencedesigner.com/blog/updating-raspbian-to-make-hubpiwi-work/5375/)

agherzan, 2016-12-22:
@mathewpeterson thank you for your feedback. We will try to include support in one of the next releases.

mathewpeterson, 2016-12-22:
Awesome, thank you for the quick reply! 

Is there anything I can do in the mean time to get this to work?

Sorry if this is not the correct place to ask these questions.

agherzan, 2016-12-22:
Definitely you can try a PR which includes the needed firmware for the device you pointed too (and maybe some kernel configuration too).

mathewpeterson, 2017-01-18:
Hello @agherzan sorry, I have not been successful with this. 

Do you have any ETA for this? This is preventing me from using Resin for about 5 devices right now.

agherzan, 2017-01-18:
@mathewpeterson I just added this supported in morty flavor of resin. We will switch to morty pretty soon so this will bring in your firmware too. 

https://github.com/resin-os/meta-resin/pull/545

agherzan, 2017-01-18:
https://github.com/resin-os/meta-resin/pull/545

agherzan, 2017-01-18:
Sorry for the above message. It was for another wifi device support . We haven't looked into these wifi chipsets yet. And currently don't have any estimation on when will that arrive. We will post here as soon as that happens.

ZubairLK, 2020-01-27:
Closing old/stale issues for cleanup.

closed 2020-01-27 by ZubairLK