balena-os/meta-balena #72: When no config.json is present, /mnt/conf/config.json directory is created

agherzan, 2016-05-18:
There seems to be a bug in our scripts cause when the config.json file doesn't exist, a strange directory is created with the same name.

brownjohnf, 2016-05-18:
Is it possible this is caused by attempting to mount that non-existent file into a container? I believe that would cause a directory to be created with the name of the file. I've seen this happen in the dev env when attempting to start the supervisor without a config.json file.

agherzan, 2016-05-18:
Yeah... Sounds like it. Good hint. Want to try a patch on it? @brownjohnf

brownjohnf, 2016-05-18:
Sure! I may need a little guidance about where to look.

agherzan, 2016-05-19:
I can point you where we start the supervisor: https://github.com/resin-os/meta-resin/blob/master/meta-resin-common/recipes-support/resin-supervisor/files/resin-supervisor.service . This service runs the container and uses:
-v $CONFIG_PATH:/boot/config.json

Config path comes from: resin-vars

agherzan, 2017-07-18:
config.json is now created by the build system so it will always be in place. Closing this issue.

closed 2017-07-18 by agherzan