balena-os/meta-balena #92: [Suggestion] Submit layer to OpenEmbedded Metadata Index

charleswhchan, 2016-06-05:
Typically, one uses the OE index to search for functionality, kind of like the "Google for layers" :smile:. (took me a while to found out resin.io works for Yocto)
https://layers.openembedded.org/layerindex/branch/master/layers/

To improve visibility of this probject, I suggestion your organization to submit the information to the index so others can find it more easily.
https://layers.openembedded.org/layerindex/submit/

agherzan, 2016-06-06:
That is something we want to do in the near future. Currently we haven't tested meta-resin on its own. And we want to make it generic bringing complete functionality by its own. In the current setup meta-resin is used with/in resin-* repositories and has a couple of dependencies there. Will keep you updated when we will have this ready. Thanks for the suggestion.

charleswhchan, 2016-06-06:
@agherzan, thanks for the explanation. I will keep an eye on it.

agherzan, 2016-07-15:
Sent entries for all the meta-resin* layers. Waiting for them to be published.

agherzan, 2016-07-19:
All the entries were approved on http://layers.openembedded.org/layerindex/branch/master/layers/ . Search 'resin' to get all the published layers.

closed 2016-07-19 by agherzan