balena-os/meta-balena #98: Monitor core services and report fails to API

agherzan, 2016-06-07:
Let host monitor a couple of core services (VPN, rce, connman). If any of these are down, send progress message: "Something is wrong on your device. Contact support." Or so.

agherzan, 2016-06-07:
This needs to be postponed as there are some design changes to be done.

alexgg, 2020-04-17:
This is something for device-health to handle. Currently we run the checks on a user request basis, maybe async API alarms are also to consider. @xginn8 do you know if this is part of device health roadmap so it can be closed in meta-balena until it's better defined?

xginn8, 2020-04-20:
@alexgg this'll probably be best encapsulated in device monitoring (rather than self-healing), likely as part of multiapp.

alexgg, 2020-05-04:
Thanks @xginn8 I am closing this as this seems to be a placeholder for something bigger.

closed 2020-05-04 by alexgg