balena-os/meta-balena #99: Set hostname in host OS

agherzan, 2016-06-07:
the yocto host OS has its hostname set to device-type then the supervisor starts a container with host networking, which doesn't create a virtual ethernet device and the container gets the default value, which is the same as the host.
After that the hostname is set again by base-images (<device-type>-<device-uuid>)

We want move this to the yocto host OS so we don't need to do that in our base-images.
This is the correct format: <device-type>-<device-uuid> (6 digits uuid only)

willnewton, 2017-07-18:
@agherzan I believe this is done, ok to close?

agherzan, 2017-07-18:
It was implemented here: https://github.com/resin-os/meta-resin/commit/b1bbf71777a59801f0b427b7bcafcf4c4ca55a54 . In the current implementation we decided to use UUID as default hostname. So I'll close this ticket for now. If this format needs to be changed, we can reopen it.

closed 2017-07-18 by agherzan