balena-os/meta-balena #108: Normal provisioning without deltas

agherzan, 2016-06-08:
# Description 
Provision a device and check that supervisor works well without and with deltas

# Steps to reproduce issue 
Provision a board on a test application. Ensure RESIN_SUPERVISOR_DELTA config var is *not* set to "1".
Push a simple application (e.g. a loop that prints to console).
Once updated, test LED blink and Web Terminal.
Set an environment variable FOO=bar.

# Expected result 
Device should appear on the dashboard.
Device should download the application and run it.
LED should blink.
Web Terminal should open and allow running commands.
When the env var is set, the device should update the application to run with the new var. Opening a web terminal and running "env" should show the variable is correctly set.

# Other information 
Reported by: Shaun Mulligan 

Test configuration: Artik 10 

Test case number: TC09 

Test run: [https://resinio.testlodge.com/projects/16238/runs/212560?tab=2&run_section_id=210889&executed_case_id=10046749#executed_case_10046749](https://resinio.testlodge.com/projects/16238/runs/212560?tab=2&run_section_id=210889&executed_case_id=10046749#executed_case_10046749)

floion, 2016-06-09:
@shaunmulligan , some fixes have been done. Can you verify with latest image this and close it if it is fixed?

agherzan, 2016-06-13:
Confirmed with @shaunmulligan that we can close this.

closed 2016-06-13 by agherzan