balena-os/meta-balena #109: Adding deltas to a running supervisor

agherzan, 2016-06-08:
# Description 
A supervisor that is running without deltas starts using them.

# Steps to reproduce issue 
Use the board from TC09, without reprovisioning.
Add an *application-wide* RESIN_SUPERVISOR_DELTA = 1 config var. (It is important that the value is 1).
Push an update to the application (e.g. changing what is outputted to the console).

# Expected result 
When setting RESIN_SUPERVISOR_DELTA, the device  should *not* restart the app.
The device should download the update correctly after pushing.
Opening a web terminal and running "env" should show that the config env var is correctly set (RESIN_SUPERVISOR_DELTA=1).

# Other information 
Reported by: Shaun Mulligan 

Test configuration: Artik 10 

Test case number: TC10 

Test run: [https://resinio.testlodge.com/projects/16238/runs/212560?tab=2&run_section_id=210889&executed_case_id=10046750#executed_case_10046750](https://resinio.testlodge.com/projects/16238/runs/212560?tab=2&run_section_id=210889&executed_case_id=10046750#executed_case_10046750)

floion, 2016-06-09:
CC'ing @petrosagg

floion, 2016-06-09:
also @agherzan

floion, 2016-06-09:
@shaunmulligan , some fixes have been done. Can you verify with latest image this and close it if it is fixed?

floion, 2016-06-28:
Fixed

closed 2016-06-28 by floion