balena-os/meta-balena #110: Provisioning a device with deltas already enabled

agherzan, 2016-06-08:
# Description 
Provisioning a device on an app that already has deltas pushed

# Steps to reproduce issue 
Provision a device on the same app used for TC10 (that already has the RESIN_SUPERVISOR_DELTA=1 var and a pushed image).

# Expected result 
The device should appear on the dashboard and correctly download the image for the app. Running a web terminal and running "env" should show the  RESIN_SUPERVISOR_DELTA env var set to 1.

# Other information 
Reported by: Shaun Mulligan 

Test configuration: Artik 10 

Test case number: TC11 

Test run: [https://resinio.testlodge.com/projects/16238/runs/212560?tab=2&run_section_id=210889&executed_case_id=10046751#executed_case_10046751](https://resinio.testlodge.com/projects/16238/runs/212560?tab=2&run_section_id=210889&executed_case_id=10046751#executed_case_10046751)

floion, 2016-06-09:
CC'ing @petrosagg

floion, 2016-06-09:
also @agherzan

floion, 2016-06-09:
Closing issue as it was not meta-resin related

closed 2016-06-09 by floion