balena-os/meta-balena #160: Identification LED

agherzan, 2016-06-30:
# Description 
Test that the identification LED is working

# Steps to reproduce issue 
Go to the device's web page
Click on IDENTIFY DEVICE (BLINK LED)

# Expected result 
Device should blink a LED for a couple of seconds.

# Actual result 
power LED is on, no LED is blinking

# Other information 
Reported by: Gergely Imreh 

Test configuration: Parallella 

Test case number: TC03 

Test run: [https://resinio.testlodge.com/projects/16238/runs/216494?tab=2&run_section_id=220101&executed_case_id=10168312#executed_case_10168312](https://resinio.testlodge.com/projects/16238/runs/216494?tab=2&run_section_id=220101&executed_case_id=10168312#executed_case_10168312)

imrehg, 2016-08-17:
Still the same result with Resin OS 1.8.0

willnewton, 2017-04-18:
No further development is planned for this board.

closed 2017-04-18 by willnewton