balena-os/meta-balena #162: Provision on Ethernet

agherzan, 2016-07-01:
# Description 
Provision the device using the Ethernet port.

# Steps to reproduce issue 
Download Device OS
Network Connection: Ethernet
Download OS file

# Expected result 
Device should appear in the application's dashboard

# Other information 
Reported by: Theodor Gherzan 

Test configuration: TS7700 

Test case number: TC01 

Test run: [https://resinio.testlodge.com/projects/16238/runs/216494?tab=2&run_section_id=220109&executed_case_id=10168350#executed_case_10168350](https://resinio.testlodge.com/projects/16238/runs/216494?tab=2&run_section_id=220109&executed_case_id=10168350#executed_case_10168350)

telphan, 2016-07-01:
Issue moved to [resin-os/resin-ts #26](https://github.com/resin-os/resin-ts/issues/26) via [**ZenHub**](https://www.zenhub.com/)

willnewton, 2017-04-18:
No further development is planned for this board.

closed 2017-04-18 by willnewton