balena-os/meta-balena #166: Layers with size of around 2GB cause rce/Docker to exit with an out of memory error while updating.

agherzan, 2016-07-04:


agherzan, 2018-02-02:
@imrehg is trying to reproduce this.

imrehg, 2018-02-05:
@agherzan currently testing this through the regular resin infrastructure does not work do to an open registry issue: https://github.com/resin-io/resin-registry2/issues/8

As an alternate test: trying `resin local push`, I've successfully deployed a 4000MB layer fine. Thus I would say:

* `local push` suggests that this might not be an issue anymore
* if it's still an issue, then it would be through deploying `git push` or `resin deploy`, and the registry's issue prevents such large images to reach the device at the moment anyways (ie. releases with such big layers cannot be created at the moment).

agherzan, 2018-02-05:
Thanks @imrehg for this investigation. I will close in that case this issue for now.

closed 2018-02-05 by agherzan

imrehg, 2018-02-05:
Just for reference, tried one more test based on comments by @shaunmulligan: push an image with a 2000MB layer to Docker HUB and then `balena pull` onto the device. That should be even closer to the regular deploy. Seemed to have gone through fine.