balena-os/meta-balena #175: Add post resin BSP layer creation information

agherzan, 2016-07-04:
The http://docs.resin.io/#/pages/hardware/meta-resin.md document needs information on the steps needed after the resin BSP layer was created.

We can generalize these steps in README.md (meta-resin) as these will be independent of the resin BSP layer. In that case, meta-resin.md should point to these instructions in README.md (meta-resin).

floion, 2017-01-25:
@imrehg @craig-mulligan I believe the info in http://docs.resin.io/#/pages/hardware/meta-resin.md is deprecated. The latest info on adding device support is available at https://github.com/resin-os/meta-resin/blob/master/contributing-device-support.md
@agherzan after the docs are updated, is there anything else to be done in this issue?

agherzan, 2017-01-25:
@floion we need to figure out what is needed after the image is created in yocto. Basically how do you hook the board support process in the resin infrastructure and provide this information in the documentation.