balena-os/meta-balena #213: Inconsistency in systemd unit naming

lorenzo-stoakes, 2016-07-22:
Currently we name our systemd services located in `/lib/systemd/system/` rather inconsistently. Many have the `resin-` prefix but others do not, most notably `docker.service` which is hard to tell apart from a default one, `mnt-data.service` is another prime example.

It can be useful when debugging an issue to see which services are ours in order to run `systemctl status`, indeed sometimes you want to determine what docker, openvpn the supervisor etc. are up to and which dependencies are ours, and this lack of consistency is made more difficult than it needs to be.

Switching to simply prefixing everything of our's to `resin-` should be a pretty quick fix.

shaunmulligan, 2016-10-05:
the openvpn-resin service is especially inconsistent and annoying when digging around in the device :/

agherzan, 2018-02-02:
From this issue I will take the openvpn-resin service.

All the other examples are either upstream components (not resin specific), for example docker/balena, or systemd naming conventions (mount services).

commit 3fe4c44b (2018-02-07): resin-mount: Rename partition services and fix service start/stop

After merging the in container resinOS PR
(https://github.com/resin-os/meta-resin/pull/951) we broke mounting the state
partition on boards. This is due to
https://github.com/resin-os/meta-resin/issues/975 . This change fixes the issue
above by adding a wrapper which checks if a certain label is already mounted.
It handles as well umounts so we can stop the partition services as well. These
services were updated to use this mount wrapper (reisn-partition-mounter).

The second change this introduces is the renaming from mnt-(lab [... cut]
commit 55eed7a4 (2018-02-07): resin-mount: Rename partition services and fix service start/stop

After merging the in container resinOS PR
(https://github.com/resin-os/meta-resin/pull/951) we broke mounting the state
partition on boards. This is due to
https://github.com/resin-os/meta-resin/issues/975 . This change fixes the issue
above by adding a wrapper which checks if a certain label is already mounted.
It handles as well umounts so we can stop the partition services as well. These
services were updated to use this mount wrapper (reisn-partition-mounter).

The second change this introduces is the renaming from mnt-(lab [... cut]
commit 7c05ac89 (2018-02-08): resin-mount: Rename partition services and fix service start/stop

After merging the in container resinOS PR
(https://github.com/resin-os/meta-resin/pull/951) we broke mounting the state
partition on boards. This is due to
https://github.com/resin-os/meta-resin/issues/975 . This change fixes the issue
above by adding a wrapper which checks if a certain label is already mounted.
It handles as well umounts so we can stop the partition services as well. These
services were updated to use this mount wrapper (reisn-partition-mounter).

The second change this introduces is the renaming from mnt-(*). [... cut]
commit 06befb50 (2018-02-09): resin-mount: Rename partition services and fix service start/stop

After merging the in container resinOS PR
(https://github.com/resin-os/meta-resin/pull/951) we broke mounting the state
partition on boards. This is due to
https://github.com/resin-os/meta-resin/issues/975 . This change fixes the issue
above by adding a wrapper which checks if a certain label is already mounted.
It handles as well umounts so we can stop the partition services as well. These
services were updated to use this mount wrapper (reisn-partition-mounter).

The second change this introduces is the renaming from mnt-(*). [... cut]
closed 2018-02-09 by agherzan