balena-os/meta-balena #214: wpa_supplicant doesn't support DBus introspection

petrosagg, 2016-07-23:
The default configuration of openembedded ships wpa-supplicant without DBus introspection[1] but that makes interacting with it very awkward because some DBus libraries are based on introspection to provide their APIs. `node-dbus` is one of them.

Unless there is good reason against it, I think it's a good idea to enable `CONFIG_CTRL_IFACE_DBUS_INTRO`

http://cgit.openembedded.org/openembedded-core/tree/meta/recipes-connectivity/wpa-supplicant/wpa-supplicant/defconfig?h=jethro#n390

agherzan, 2016-08-23:
@petrosagg Do we still need this? As far as I remember, I heard you saying that it was needed for something that you changed in the meanwhile. Can you please confirm and close this issue if so?

petrosagg, 2016-08-26:
We don't need it but it'd be nice to have. I'll close it now till the next time we hit it if ever.

closed 2016-08-26 by petrosagg