balena-os/meta-balena #226: upx doesn't support aarch64 binary format

agherzan, 2016-08-01:
|                        Ultimate Packer for eXecutables
|                           Copyright (C) 1996 - 2013
| UPX 3.91        Markus Oberhumer, Laszlo Molnar & John Reiser   Sep 30th 2013
| 
|         File size         Ratio      Format      Name
|    --------------------   ------   -----------   -----------
| upx: <...>/resin-provision: UnknownExecutableFormatException

willnewton, 2017-04-18:
This has been worked around by disabling UPX compression on aarch64 for now.

agherzan, 2017-07-18:
Closing this issue as we are not using UPX anymore. The only place where we still have it is in the provisioner which will get rewritten completely. As well, in the meanwhile, the new version of UPX does support aarch64.

closed 2017-07-18 by agherzan