balena-os/meta-balena #240: Update available RPi overlays

imrehg, 2016-08-14:
Update available overlays to match the ones in the Raspberry Pi [firmware repo](https://github.com/raspberrypi/firmware/tree/master/boot/overlays). Since it is not really possible to add new device overlays at the moment remotely (they live in the host, so have to have access to the SD card to add), only enable/disable them, it would be great to have all of them available for users. This also removes one possible surprise / discrepancy compared to using Raspberry Pi with its own OS outside of resin.

imrehg, 2016-09-01:
I've checked and we have very few overlays (15) compared to the firmware repo. Including the new ones would be very beneficial.

Some example: a lot of people (including us) are using PiTFT 2.8" displays. Currently they are done through kernel drivers, but actually they have workable overlays that would simplify setup, and likely also make the display available before the applications start (e.g. splash screens).

This might also need a kernel version update.

agherzan, 2017-07-18:
All overlays were updated and included in resinOS when the BSP was updated. Closing this issue.

closed 2017-07-18 by agherzan