balena-os/meta-balena #248: Push simple python application

agherzan, 2016-08-17:
# Description 
Push an a simple python application to the device

# Steps to reproduce issue 
git clone https://github.com/alexandrosm/hello-python
cd hello-python
git remote add resin YYY (copy the link from the application's dashboard)
git push resin master

# Expected result 
In the device's logs you should see: "hello python!"

# Actual result 
Can't download application

# Other information 
Reported by: Gergely Imreh 

Test configuration: QEMU X86 32bit 

Test case number: TC06 

Test run: [https://resinio.testlodge.com/projects/16238/runs/221686?tab=2&run_section_id=231829&executed_case_id=10347326#executed_case_10347326](https://resinio.testlodge.com/projects/16238/runs/221686?tab=2&run_section_id=231829&executed_case_id=10347326#executed_case_10347326)

jnv, 2016-09-04:
As a workaround, you can resize the image using `qemu-img` and Resin automatically resizes the partition, e.g.:

```
qemu-img resize -f raw resin-image-qemux86-64.img +4G
```

#249 is duplicate.

agherzan, 2016-09-05:
CC @lifeeth i think we should have this instruction in qemu machines. Cause my default docker disk is only 1G which might be insufficient for some apps.

agherzan, 2016-09-05:
Or... I would actually vote to start shipping qemu images with 4G of docker disk (most of the boards we have use that disk size). CC @petrosagg

petrosagg, 2016-09-05:
SGTM

agherzan, 2018-02-06:
This issue was moved to resin-os/resin-qemu#82

closed 2018-02-06 by agherzan