balena-os/meta-balena #250: Sync application container

agherzan, 2016-08-17:
# Description 
Update code on a provisioned device using Resin CLI _`resin sync`_ command

# Steps to reproduce issue 
1. Install resin cli with _`npm i -g resin-cli`_
2. Login with _`resin login`_. If testing on staging, prepend RESINRC_RESIN_URL=resinstaging.io to this and all subsequent `resin` commands.
3. Clone [this test node js application](https://github.com/resin-io-projects/simple-server-node) on your laptop and provision a device with it. 
4. Wait until the application is marked as running on the dashboard.
5. Enable and navigate to the Public Device URL. Confirm that the web server shows a `Hello World` message.
6. Edit `server.js` of the cloned application so that `res.send()` returns a different message, e.g. `res.send('Hello Resin Sync!')`
7. Run _`resin devices`_ to get the device uuid of the provisioned device.
8. Ensure that your resin account has at least one RSA or DSA key -preferably RSA- that is also loaded in your ssh-agent. Use `ssh-add -l` to check the loaded keys.
8. Run _`time resin sync <short device uuid>`_ within the test application folder (i.e. the folder of the edited `server.js` file). For staging run _`time RESINRC_RESIN_URL=resinstaging.io resin sync <short device uuid>`_

# Expected result 
Expected result

* The sync process should start with status messages appearing on each step ('Stopping application container...', 'Syncing..', 'Starting application container...').
* At the end of _`resin sync`_ a "resin sync completed successfully!" message should appear.
* The device's Public Device URL should now show the new message, e.g. 'Hello Resin Sync!'

# Actual result 
resin sync completes, but copies the files to the wrong place. The new if I search for `server.js`, I find it at `./. ./server.js`, wherever that might be actually....

Here's the transcript (with a bit of snip to show only relevant info)

```
greg@nostromo:simple-server-node 18:08:06 (0) <528>
$RESINRC_RESIN_URL=resinstaging.io resin sync ee764be6931fdebd3fe7c56423ddaed42e93d16029a1968cb14806451ecf26
Connecting with: ee764be6931fdebd3fe7c56423ddaed42e93d16029a1968cb14806451ecf26
Application container stopped.
Synced /usr/src/app on ee764be.
Application container started.

resin sync completed successfully!
greg@nostromo:simple-server-node 18:09:01 (0) <529>
$RESINRC_RESIN_URL=resinstaging.io resin ssh ee764be6931fdebd3fe7c56423ddaed42e93d16029a1968cb14806451ecf26
Connecting with: ee764be6931fdebd3fe7c56423ddaed42e93d16029a1968cb14806451ecf26
root@parallella-hdmi-resin-ee764be:/usr/src/app# cd /
root@parallella-hdmi-resin-ee764be:/# find . -name "server.js"
./usr/src/app/server.js
./. ./server.js
```

# Other information 
Reported by: Gergely Imreh 

Test configuration: Parallella 

Test case number: TC14 

Test run: [https://resinio.testlodge.com/projects/16238/runs/221686?tab=2&run_section_id=231835&executed_case_id=10347339#executed_case_10347339](https://resinio.testlodge.com/projects/16238/runs/221686?tab=2&run_section_id=231835&executed_case_id=10347339#executed_case_10347339)

lifeeth, 2016-08-17:
Just dropped by to say that this issue is nicely filed.

imrehg, 2016-08-17:
@lifeeth big thanks for those who made the template :)

willnewton, 2017-04-18:
No further development is planned for this board.

closed 2017-04-18 by willnewton