balena-os/meta-balena #257: Compress kernel modules for all the boards

floion, 2016-08-19:


imrehg, 2016-08-22:
Is there a list of board where this is enabled/disabled?

floion, 2016-08-22:
The is no list. But at this moment we only have this for the rpi's and the beaglebone

agherzan, 2016-08-23:
@petrosagg How are we going to fix the debian images?

agherzan, 2016-08-25:
We should do this for consistency purposes but we can consider this as a low priority task.

lurch, 2016-10-14:
I guess this is what #303 is about?

lurch, 2016-10-18:
https://github.com/resin-os/resinos/issues/89 suggests that #257 and #303 should be closed :-/

agherzan, 2016-10-18:
Closing this because we encountered some issue with debian based images which don't have support for compressed kernel modules. Fixing this involves hacks. In the meanwhile we enlarged the rootfs partition to cope with these space requirements.

closed 2016-10-18 by agherzan