balena-os/meta-balena #268: resinOS as an OS base for IoT

jgato, 2016-08-30:
Hi there,

Sorry if this is not the right place to do this kind of questions. Is it possible (or has sense) to build a resinOS to create a base OS for IoT devices? To create images like for example RancherOS.

This base image would not necessary needs to connect to resin services (I have seen in the doc this would be possible), you just have a base OS which is really good about running docker containers. Because, I can see that you are maybe one of the most advanced people about running containers in many devices. I am finding a lot of difficulties using RancherOS in the Intel Edison, or even Raspberry, but you seem to support many many devices.

agherzan, 2016-08-30:
Hi @jgato . We do indeed have a mechanism to build images for people who only want to run docker on their target. In this direction we define a variable called `RESIN_CONNECTABLE`. Based on this variable, the image created by Yocto will contain resin specific bits or will be a plain yocto image which will only be intended to run docker. You can as well preload a docker image with some additional variables. The entire documentation is available here: https://github.com/resin-os/meta-resin/blob/master/README.md#resin_connectable--0 . 

Regards,
Andrei

willnewton, 2017-04-18:
I'm closing this issue as it seems to be resolved. If you have any further questions please feel free to start a new topic at https://forums.resin.io where we will be happy to help you with any problems you might have. Thanks!

closed 2017-04-18 by willnewton