balena-os/meta-balena #280: The name "/resin_supervisor" is already in use by container <container_hash>. You have to remove (or rename) that container to be able to reuse that name..

floion, 2016-09-06:
docker version
Client:
 Version:      1.10.3
 API version:  1.22
 Go version:   go1.6
 Git commit:   20f81dd-unsupported
 Built:        Mon Sep  5 12:25:41 2016
 OS/Arch:      linux/arm

Server:
 Version:      1.10.3
 API version:  1.22
 Go version:   go1.6
 Git commit:   20f81dd-unsupported
 Built:        Mon Sep  5 12:25:41 2016
 OS/Arch:      linux/arm

And the actual error is:

Sep 04 16:23:16 ts-imx6 systemd[1]: resin-supervisor.service: Main process exited, code=exited, status=125/n/a
Sep 04 16:23:17 ts-imx6 docker[11899]: Failed to stop container (resin_supervisor): Error response from daemon: No such container: resin_supervisor
Sep 04 16:23:18 ts-imx6 docker[11907]: Failed to remove container (resin_supervisor): Error response from daemon: No such container: resin_supervisor
Sep 04 16:23:18 ts-imx6 systemd[1]: resin-supervisor.service: Unit entered failed state.
Sep 04 16:23:18 ts-imx6 systemd[1]: resin-supervisor.service: Failed with result 'exit-code'.
Sep 04 16:23:28 ts-imx6 systemd[1]: resin-supervisor.service: Service hold-off time over, scheduling restart.
Sep 04 16:23:28 ts-imx6 systemd[1]: Stopped Resin supervisor.
Sep 04 16:23:28 ts-imx6 systemd[1]: Starting Resin supervisor...
Sep 04 16:23:29 ts-imx6 docker[11924]: Failed to stop container (resin_supervisor): Error response from daemon: No such container: resin_supervisor
Sep 04 16:23:30 ts-imx6 docker[11951]: Failed to remove container (resin_supervisor): Error response from daemon: No such container: resin_supervisor
Sep 04 16:23:30 ts-imx6 systemd[1]: Started Resin supervisor.
Sep 04 16:23:31 ts-imx6 bash[11959]: docker: Error response from daemon: Conflict. The name "/resin_supervisor" is already in use by container 45e7b1d1b444bb0719bb3d2fb08d4d0dba85d54981dd42184fdf08f742192c16. You have to remove (or rename) that container to be able to reuse that name..
Sep 04 16:23:31 ts-imx6 bash[11959]: See '/usr/bin/docker run --help'.

floion, 2016-09-06:
Issue present also here: https://github.com/docker/docker/issues/23371

agherzan, 2017-07-18:
This was mitigated by not removing the container with every start.

closed 2017-07-18 by agherzan