balena-os/meta-balena #305: Create /resin-conf/lib/firmware/updates/ directory and bind mount it to /lib/firmware/updates/

floion, 2016-09-21:
This is done so that the supervisor can then bind mount /lib/firmware/updates/ in the user container and users can use this read-write directory to put firmware files in it.  (this is needed in the context of read-only root filesystem coupled with the fact that users may need additional firmware to add)

imrehg, 2016-09-21:
Would this enable users adding new firmware through installing in the `Dockerfile`? In the current implementation of bind mount, app container firmware directory will be mounted over. So could not instal firmware in a `RUN apt-get install <some-firmware-package>`, but need to copy/install firmware in the startup script, run every time the container starts. This is not ideal. Also not ideal, that firmware copied are left around. It should live only as long as the application lives, imho.

floion, 2016-09-21:
apt-get'ing a firmware package where does it usually install the firmware into? Is it not into /lib/firmware?

imrehg, 2016-09-21:
@floion yes, it installs there. And currently that does not work, as the bind mount overrides/hides the contents of the container `/lib/firmware` with the host's bind mounted firmware directory.

floion, 2016-09-22:
There would be another way that would save us the copy and allow for apt-get'ing in the user container. We could have the supervisor start the user container with a volume (not a bind mounted host volume, just a regular volume) pointing at /usr/firmware/ in the container. Having this, we would have a corresponding folder in the host Os under /var/lib/docker/volumes/<some_sha>. From the host Os we could then use "docker inspect" to get the exact location and we could then use that to mount it over the host Os mountspace in /lib/firmware/updates/. In this the firmware an user downloads in the container is made available in the host OS for actual loading.
@petrosagg @imrehg any thoughts?

floion, 2016-09-27:
ping @petrosagg ?

petrosagg, 2016-10-03:
I don't think we can avoid copying the firmware at runtime because the kernel will attempt to load the firmware from the host mount namespace anyway. So even if we didn't shadow `/lib/firmware` in the container you'd still have a problem. So for now I think bind-mounting `/lib/firmware/updates` and letting the container copy files there  is the best option

ZubairLK, 2020-01-27:
Closing issue to cleanup. Can reopen if still relevant/needed

closed 2020-01-27 by ZubairLK

sergbbb, 2020-01-30:
Hello, where i shall to create /resin-conf/lib/firmware/updates/ ? on first partition  of image ? Thanks!

sergbbb, 2020-02-01:
> This is done so that the supervisor can then bind mount /lib/firmware/updates/ in the user container and users can use this read-write directory to put firmware files in it. (this is needed in the [...]

Hello, where i shall to create /resin-conf/lib/firmware/updates/ ? on first partition of image ? Thanks!