balena-os/meta-balena #313: Enter running application container

agherzan, 2016-09-23:
# Description 
Get a shell into a running application container using Resin CLI _`resin ssh`_ command.

# Steps to reproduce issue 
1. Install resin cli with _`npm i -g resin-cli`_
2. Login with _`resin login`_. If testing on staging, prepend `RESINRC_RESIN_URL=resinstaging.io ` to this and all subsequent _`resin`_ commands.
3. Create a single-container test application and provision a device with it.
4. Wait until the application is marked as running on the dashboard.
5. Run _`resin devices`_ to get the device uuid.
6. Ensure that your resin account has at least one RSA or DSA key -preferably RSA- that is also loaded in your ssh-agent. Use `ssh-add -l` to check the loaded keys.
7. Run _`time resin ssh <short device uuid>`_. For staging run _`time RESINRC_RESIN_URL=resinstaging.io resin ssh <short device uuid>`_

# Expected result 
* A shell prompt (e.g. `bash-4.3#`) should appear after a few seconds.
* Running _'env | grep RESIN'_ should return a list of environment variables.

# Actual result 
Cannot connect:

$RESINRC_RESIN_URL=resinstaging.io resin ssh d6a0c17
Connecting with: d6a0c17
ResinRequestError: Request error: tunneling socket could not be established, cause=socket hang up

Web terminal does not work either

# Other information 
Reported by: Gergely Imreh 

Test configuration: Parallella 

Test case number: TC13 

Test run: [https://resinio.testlodge.com/projects/16238/runs/229418?tab=2&run_section_id=250857&executed_case_id=10623943#executed_case_10623943](https://resinio.testlodge.com/projects/16238/runs/229418?tab=2&run_section_id=250857&executed_case_id=10623943#executed_case_10623943)

imrehg, 2016-09-23:
Naturally, same issue with `resin sync`

floion, 2016-09-23:
@lekkas can you investigate this?

willnewton, 2017-04-18:
No further development is planned for this board.

closed 2017-04-18 by willnewton