balena-os/meta-balena #317: Install avahi

agherzan, 2016-09-23:
Avahi service will be installed and enabled on development and normal resin-image. This service needs to be able resolve local hostname using `libnss-mdns`.

This needs a test case.

imrehg, 2016-09-24:
This is great! One difficulty, that devices of the same type have the same host name set at the moment, isn't it? Would that cause issue with the avahi operation, e.g. having a bunch hosts called `raspberrypi3` on the network?

agherzan, 2016-09-29:
@imrehg By default each device will have uuid as hostname. The preconfigured `resin` one will be used only in development where it is expected to have only one device in the local network. See the referenced #318

imrehg, 2016-09-30:
thanks, replying there

closed 2016-10-04 by agherzan