balena-os/meta-balena #318: Configure hostname based on a config.json attribute

agherzan, 2016-09-23:
Hostname will be set to `hostname` in config.json if available otherwise will use the generated `uuid`.

imrehg, 2016-09-30:
@agherzan is that the full uuid or the shortened (7 digit) version? Because the full uuid is harsh to work with. I guess router UIs would be hard to handle it, or any case when copy-paste is not an option.

I'm not so sure that getting rid of the device type identification is great, but can make the case either way (minus: harder to group devices, plus not "leaking" device type info through hostnames)

With `config.json` modifications, this feels like a complication on the process, I guess the flexibility is worth it, but only if the defaults are sensible. Might worth asking people how do they use the host names if they do? (gitter, twitter, suchalike).

agherzan, 2016-10-05:
Sounds right @imrehg . What I'm going to do is create another issue as the connected PR was already merged and track your proposal later. Thanks.

agherzan, 2016-10-05:
For reference: https://github.com/resin-os/meta-resin/issues/364

closed 2016-10-05 by agherzan