balena-os/meta-balena #371: Resin-hostname lacks atomicity

telphan, 2016-10-06:
Due to the introduction of read-only rootfs hostnamectl cannot be used to set the static hostname value, thus we rely on manual non atomic ways of modifying the hostname.

agherzan, 2017-03-23:
@telphan dp we have any updates on this? Could you describe a little more what is the current solution?

telphan, 2017-05-09:
@agherzan No updates unfortunately, I have not managed to look into this. The solution I had in mind at the time is to force hostnamectl to work on a read-only rootfs, but now that we have an initramfs I am thinking we can squeeze this in there.

discordianfish, 2018-01-21:
Is there really no way to set the hostname currently without to reflashing the device? I've tried the change config.json by mounting it on the device but that doesn't seem to work.
I can't use `hostname` because I need the dhcp client to send the correct hostname.

agherzan, 2018-01-22:
The latest supervisor version (v6.6.0?) includes a feature where you can change the hostname using a variable. @pcarranzav can you confirm and detail this?

But as well, changing the hostname in config.json should work. How did you mount the boot partition?

agherzan, 2018-02-02:
The hostname file is updated in a non atomic way using - https://github.com/resin-os/meta-resin/blob/master/meta-resin-common/recipes-support/resin-hostname/resin-hostname/resin-hostname#L20 . This is though not a problem because it is generated at every boot.

closed 2018-02-02 by agherzan