balena-os/meta-balena #386: Shutdown services errors

agherzan, 2016-10-08:
```
[   82.505954] systemd[1]: Requested transaction contradicts existing jobs: Transaction is destructive.
[   82.515703] systemd[1]: systemd-journald.socket: Failed to queue service startup job (Maybe the service file is missing or not a non-template unit?): Transaction is destructive.
[   82.532574] systemd[1]: systemd-journald.socket: Unit entered failed state.
```

willnewton, 2017-07-18:
@agherzan do you know what steps are required to reproduce this?

agherzan, 2017-07-18:
This was fixed by the persistent logging re-implementation - https://github.com/resin-os/meta-resin/pull/554 . You can confirm this by booting a board and checking the boot messages for any errors like the ones reported above.

agherzan, 2017-07-18:
Actually this was reproducible at shutdown - not at boot.

closed 2017-07-18 by agherzan