balena-os/meta-balena #416: Full Atomic Host Updates

shaunmulligan, 2016-10-10:
Complete working solution with [resinHUP](https://github.com/resin-os/resinhup)

agherzan, 2018-02-02:
Currently hostapps provide the ability to do host os updates in an atomic way as long as everything is in the rootfs partition. All the changes to the boot partition or on raw disk are not atomic but for updates that involve only rootfs, are atomic.

@shaunmulligan feel free to add or reopen this issue but I'll close it for now.

closed 2018-02-02 by agherzan

imrehg, 2018-02-04:
@agherzan do I remember that we had plans to move everything into the root partition? Is that something that's possible as per our current understanding?

agherzan, 2018-02-05:
Sure. That will be handle on a per machine basis. The bsp needs to comply with how these are installed.