balena-os/meta-balena #418: Add Embedded Container Toolkit Layer

shaunmulligan, 2016-10-10:
The Embedded Container Toolkit (ECT) is a collection of tools and utilities built for running linux containers on embedded systems. 

* Adds support for a pull with docker-progress
* Is adapted to run in multiple architectures
* Supports syncing data into a container (resin-sync)
* Supports pulling with deltas (by specifying a deltaFrom parameter to the pull command)
* Supports Bootable containers (i.e. host OS as a container)
* Supports per-device configuration over signed images
* Supports strategies to transition from one container to another
* Supports update locks

agherzan, 2018-02-02:
@shaunmulligan what is this the status of this? It looks to me that it should be in the resinos repository (if at all).

shaunmulligan, 2018-02-05:
@agherzan I think we can close this, as lots of the embedded toolkit will probably be handled by balena

closed 2018-02-05 by agherzan