balena-os/meta-balena #420: enable zRAM

shaunmulligan, 2016-10-11:


willnewton, 2017-07-18:
@shaunmulligan is this still required? Where did the requirement come from?

shaunmulligan, 2017-07-18:
@willnewton its not required, but would be nice to know if it would give us any visible improvements. At the time I had read that some of the OSes for low spec machines like ChromeOS had enabled it, so thought it would be worth looking into. Do you have any experience with it?

landswellsong, 2017-09-15:
@willnewton I need zRam for a project which will either be based on resinOS or at least closely resembling. Should I do a PR when I have it working?

imrehg, 2017-09-16:
@landswellsong can you give us a bit more context on this, how it is contributing to your requirements? This would help us prioritize things, and make sure everything fits well together. Thanks!

landswellsong, 2017-09-16:
@imrehg we run multiple containers on one [possibly memory starved] device. It [usually] doesn't have a swap because everything is either SD or eMMC and the containers that are run are community-provided thus not guaranteed to be careful with memory allocations. Having zRam would give a huge asset for dealing with stale memory.

willnewton, 2017-09-18:
@landswellsong I opened a PR for adding the appropriate kernel modules here: https://github.com/resin-os/meta-resin/pull/851

I'm not sure there is a way we can have it running generically as the size of the disk is quite hardware and workload specific.

splitice, 2017-10-30:
zram as per #851 is currently merged but the module is not enabled by default, nor is any zram swap added.

Although most devices have emmc storage it might be also worth adding support for a small swap file. This can be especially useful with debug builds of applications that require additional resources.

agherzan, 2017-10-30:
@splitice Currently the idea is that resinOS brings in the kernel module and offers the possibility for the user to set zram swap from the application/container. Isn't that covering your use-case? If not, could you detail?

splitice, 2017-10-30:
@agherzan sorry I was not aware that the permissions were relaxed that far. I will certainly try that.

agherzan, 2017-10-30:
I will close this ticket for now. Do come back if you need any support @splitice .

closed 2017-10-30 by agherzan