balena-os/meta-balena #421: Firmware missing for Intel 7260 wlan adapter under ResinOS 1.16.0

Johnsel, 2016-10-12:
## Problem
We had an issue today with our WLAN adapter not working, it is an Intel 7260 and we were on ResinOS 1.16.0 when the issue happened. I was able to resolve it by getting firmware that the driver was looking for onto the filesystem.

## (applicable) dmesg output:

`[    2.254392] iwlwifi 0000:02:00.0: enabling device (0000 -> 0002)                                                                                                       
[    2.255065] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-7260-13.ucode failed with error -2`

## Steps taken to resolve:
```
cd /lib/firmware
wget https://github.com/OpenELEC/iwlwifi-firmware/raw/master/firmware/iwlwifi-7260-13.ucode
rmmod iwlwifi
modprobe iwlwifi
```

agherzan, 2016-10-12:
Hello @Johnsel . Thank you for your bug report. We are going to look into this and make sure this will be included in the next releases.

lifeeth, 2016-12-08:
I confirm that this is missing - Ran into the same issue just now.

agherzan, 2017-01-18:
I just added this supported in morty flavor of resin. We will switch to morty pretty soon so this will bring in the firmware requested above. 

https://github.com/resin-os/meta-resin/pull/545

floion, 2017-05-24:
Issue fixed in latest rsinOS release. Closing

closed 2017-05-24 by floion