balena-os/meta-balena #423: Bump supervisor to v2.6.2

pcarranzav, 2016-10-12:


lurch, 2016-10-14:
This was fixed in #424

closed 2016-10-14 by agherzan

pcarranzav, 2016-10-14:
It was "fixed" but it still didn't make it to staging, so it should probably sit as "Waiting for staging" instead of being closed.
Though now there's #425 so I don't know how we'd treat this specific case. I'd rather keep this ticket in Waiting for staging so that we have clarity about what has made it to staging and production.
But then again, we might want to revisit the way we use the Waffle labels since most of the meta-resin PRs are being written with "Fixes #issue" or "Closes #issue" instead of "Connects to #issue".

lurch, 2016-10-14:
Sorry if I've messed up your workflow, feel free to reopen.

pcarranzav, 2016-10-14:
On the contrary, thanks for bringing this up :)
@agherzan is the maintainer here and knows the proper workflow, I just commented cause there seems to be this bit of a contradiction in how tickets are handled, unless there's something I'm missing.

agherzan, 2016-10-14:
That is indeed something we need to think about it. Currently the way I see it is that all things towards meta-resin would be closed as soon as implemented and then be released at the next meta-resin release. We need to talk this internally and decide on a way forward.

lurch, 2016-10-14:
...and if you haven't already seen it, see my comment here https://github.com/resin-os/meta-resin/issues/344#issuecomment-253760350