balena-os/meta-balena #450: Persistent logging change

telphan, 2016-10-25:
We are required to provide persistent logging changes without reboot this implies a heavy change in the way we implement our persistent logging.

agherzan, 2017-01-16:
@telphan we will need a spec for this fist

floion, 2017-01-27:
Trying to boot a beaglebone on resin 2.X with morty, I get the following:

[   11.034862] systemd[1]: Starting Load/Save RF Kill Switch Status...
         Starting Load/Save RF Kill Switch Status...
[   11.283936] systemd[1]: Started Load/Save RF Kill Switch Status.
[  OK  ] Started Load/Save RF Kill Switch Status.
[   11.529093] systemd[1]: Started udev Wait for Complete Device Initialization.
[  OK  ] Started udev Wait for Complete Device Initialization.
[   11.607195] systemd[1]: Starting Show Plymouth Boot Screen...
         Starting Show Plymouth Boot Screen...
[   13.215751] systemd[1]: Received SIGRTMIN+20 from PID 410 (plymouthd).
[   13.239917] systemd[1]: Started Show Plymouth Boot Screen.
[   13.285395] systemd[1]: Started Forward Password Requests to Plymouth Directory Watch.
[   13.317039] systemd[1]: Reached target Paths.
[   14.881372] systemd[1]: resin-conf-reset.service: Main process exited, code=exited, status=1/FAILURE
[   14.901221] systemd[1]: Failed to start Resin reset OS configuration.
[   14.911063] systemd[1]: Dependency failed for OpenSSL random state bind mount.
[   14.919941] systemd[1]: Dependency failed for Resin device uuid.
[   14.932033] systemd[1]: Dependency failed for Prepare OpenVPN Application On Resin.
[   14.941428] systemd[1]: Dependency failed for OpenVPN Application On Resin.
[   14.950344] systemd[1]: openvpn-resin.service: Job openvpn-resin.service/start failed with result 'dependency'.
[   14.960778] systemd[1]: prepare-openvpn.service: Job prepare-openvpn.service/start failed with result 'dependency'.
[   14.971406] systemd[1]: Dependency failed for Resin supervisor.
[   14.978766] systemd[1]: resin-supervisor.service: Job resin-supervisor.service/start failed with result 'dependency'.
[   14.989637] systemd[1]: Dependency failed for Resin Hostname Configuration.
[   14.998084] systemd[1]: Dependency failed for Avahi mDNS/DNS-SD Stack.
[   15.006044] systemd[1]: avahi-daemon.service: Job avahi-daemon.service/start failed with result 'dependency'.
[   15.016163] systemd[1]: resin-hostname.service: Job resin-hostname.service/start failed with result 'dependency'.
[   15.028993] systemd[1]: resin-device-uuid.service: Job resin-device-uuid.service/start failed with result 'dependency'.
[   15.039920] systemd[1]: home-root-.rnd.mount: Job home-root-.rnd.mount/start failed with result 'dependency'.
[   15.050018] systemd[1]: Dependency failed for NetworkManager system connections bind mount.
[   15.060046] systemd[1]: Dependency failed for Resin network configure service.
[   15.068975] systemd[1]: resin-net-config.service: Job resin-net-config.service/start failed with result 'dependency'.
[   15.079798] systemd[1]: etc-NetworkManager-system\x2dconnections.mount: Job etc-NetworkManager-system\x2dconnections.mount/start failed with result 'dependency'.
[   15.094485] systemd[1]: Dependency failed for Docker key.json bind mount.
[   15.102790] systemd[1]: Dependency failed for Docker Application Container Engine.
[   15.111866] systemd[1]: docker.service: Job docker.service/start failed with result 'dependency'.
[   15.123162] systemd[1]: etc-docker.mount: Job etc-docker.mount/start failed with result 'dependency'.
[   15.137016] systemd[1]: Dependency failed for Resin target mount point to persistent location.
[   15.161098] systemd[1]: etc-systemd-system-resin.target.wants.mount: Job etc-systemd-system-resin.target.wants.mount/start failed with result 'dependency'.
[   15.187739] systemd[1]: Dependency failed for ResinOS hostname.
[   15.206816] systemd[1]: etc-hostname.mount: Job etc-hostname.mount/start failed with result 'dependency'.
[   15.220293] systemd[1]: Dependency failed for ResinOS persistent logging location.
[   15.229728] systemd[1]: Dependency failed for Flush Journal to Persistent Storage.
[   15.240072] systemd[1]: systemd-journal-flush.service: Job systemd-journal-flush.service/start failed with result 'dependency'.
[   15.251714] systemd[1]: Dependency failed for Resin persistent logs.
[   15.259507] systemd[1]: Dependency failed for Journal Service.
[   15.266724] systemd[1]: systemd-journald.service: Job systemd-journald.service/start failed with result 'dependency'.
[   15.277876] systemd[1]: resin-logs.service: Job resin-logs.service/start failed with result 'dependency'.
[   15.287545] systemd[1]: var-log-journal.mount: Job var-log-journal.mount/start failed with result 'dependency'.
[   15.297823] systemd[1]: Dependency failed for journald.conf.d bind mount.
[   15.306763] systemd[1]: etc-systemd-journald.conf.d.mount: Job etc-systemd-journald.conf.d.mount/start failed with result 'dependency'.
[   15.319202] systemd[1]: Dependency failed for ResinOS supervisor configuration.
[   15.327898] systemd[1]: etc-resin\x2dsupervisor.mount: Job etc-resin\x2dsupervisor.mount/start failed with result 'dependency'.
[   15.339612] systemd[1]: resin-conf-reset.service: Unit entered failed state.
[   15.346746] systemd[1]: resin-conf-reset.service: Failed with result 'exit-code'.
[   15.365366] systemd[1]: Starting Resin reset OS configuration...
[   15.414755] systemd[1]: Starting Create Volatile Files and Directories...
[   15.445839] systemd[1]: Reached target Resin Target.
[   16.267051] systemd[1]: Started Create Volatile Files and Directories.
[   16.363571] systemd[1]: Starting Network Time Synchronization...
[   16.430317] systemd[1]: Starting Update UTMP about System Boot/Shutdown...
[... cut ...]
[   27.748300] ieee80211 phy0: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
[   27.760979] ieee80211 phy0: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.36
[   27.941515] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   28.148907] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   28.308346] systemd[1]: Starting WPA supplicant...
[   28.587102] systemd[1]: Started WPA supplicant.
[   28.686812] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready

agherzan, 2017-01-27:
Are you using current master HEAD? Could you investigate why `resin-conf-reset.service` is failing?

closed 2017-03-10 by agherzan