balena-os/meta-balena #472: please add dig to the resinos host image

lfarkas, 2016-11-10:
IMHO dig is such a basic tool that it'd have to the minimal host image.
thanks.
regards.

agherzan, 2016-11-10:
Hello. Thank you for your suggestion. I will tell you what currently our view about what should end un in the host os. The resinOS is a distribution that is meant for running containers as easy as possible. This means that the host os would include only things that are required for docker to run and everything else would be candidates for in container use. That's why you will find the host os pretty minimalistic.

lfarkas, 2016-11-10:
yes i agree but in a distributed large installation it'd be useful to dynamically register each host os (and all of it's interfaces) into a dns. for this purpose dig would be very useful. yes i know there are many other ways, but still this it's a so small utility which worth for it. like rhel/centos atomic host has.

alexandrosm, 2016-11-11:
@petrosagg -- thoughts?

alexandrosm, 2016-11-11:
and actually, @hedss, thoughts?

petrosagg, 2017-01-18:
@lfarkas can you give some more info on how you'll use dig to register each host os? AFAIK dig can only perform DNS lookups.

lfarkas, 2017-01-22:
if you'd like to use bind's default nsupdate procedure and wouldn't like to hardcode the zone's primary dns ip address then the best way to look it up with eg:
`dig +nocmd +nosearch +noall +answer +short soa $1 | cut -d' ' -f1`
and dns's ip:
`dig $server $1 +nocmd +nosearch +noall +answer +short | tail -1`
but of course as always everything can be done in a container and not on the host.

agherzan, 2017-01-23:
Just to understand, how do you indent to run the 'dig' util from the host?

lfarkas, 2017-01-23:
i'd like to add a networkmanager dispatcher script when a new interface come up or disappear run shell script which then call dig to determine from the hostname and interface name which domain's to update int which zone's which dns (and update the A record of the given host)

petrosagg, 2017-03-23:
@lfarkas I didn't quite understand what you intend to do. It sounds like it's doable from inside a container. We try to be very careful on which utilities we include in the host OS and try to keep the size small. Obviously `dig` on its own is not a very big difference but a lot of different utilities could be very useful to different people and then they add up.

If you plan adding dispatch scripts to networkmanager does that mean you're going to build resinOS on your own? Because if that is the case you could add dig or any other customisations in your custom build. Would that work for your usecase?

lfarkas, 2017-03-23:
i think this is no longer valid since it can be done in a container.

closed 2017-03-23 by lfarkas

agherzan, 2017-03-23:
Thanks @lfarkas