balena-os/meta-balena #478: Add option in "rdt configure" for setting SSH key

javefang, 2016-11-20:
Currently an resinos installation is open to anyone on the network to access. Is it a good idea to allow a way for specifying an SSH key? e.g. when configuring the image using "rdt configure"?

agherzan, 2016-11-28:
Is definitely something we talked about and we want. I know @hedss is building something along these lines so he might give us more details.

agherzan, 2018-02-02:
@shaunmulligan @hedss Any update on this? Is this still planned or maybe done already?

hedss, 2018-02-02:
Not that I know of. @shaunmulligan will if anyone does, though.

shaunmulligan, 2018-02-05:
@agherzan its not implemented, but I would like the resin cli to configure this. @pimterry can we add this as a feature request for `resin configure` allowing the user to specify an SSH key.

pimterry, 2018-02-05:
@shaunmulligan how does the CLI inject the SSH key into the image? Is there a way to do that using just the boot partition? That needs to exist first, as right now the CLI can't access the root of any new (>2.7.5) devices, because there's no filesystem drivers that can read the hostapp partition format. I expect that'll come as part of reconfix, but that's still a way away.

hedss, 2018-02-05:
@pimterry Maybe some of the work for the `import-images` tool could be ripped out and used instead? This injects device keys into images: https://github.com/resin-io/resin-containers/blob/master/tools/import-images/src/slugWrangler.coffee#L279

I don't know (but assume) we're using the same mechanism for Dropbear, with authenticated keys, but that the root pw is squashed by default? I'm happy to discuss this/look into the mechanism. :)

shaunmulligan, 2018-02-05:
@pimterry I think we will need @agherzan to implement something similar to what we have for `system-connections`. So we could have a file or folder called `authorized_keys` and that could exist in the boot partition. I believe this is already scheduled to be part of the OS unification stuff and there are at least two users who have implemented something along these lines themselves

pimterry, 2018-02-05:
There's definitely useful stuff in there for this @hedss (I think that's actually got some of the best documentation on how to some partition details I've seen), but it relies on `mount` which we can't do in the CLI, we need a JS filesystem driver, and I'm fairly sure that our current [`node-ext2fs`](https://github.com/resin-io/node-ext2fs) doesn't easily work. That code does suggest the formats are closer than I'd thought though, so there might be a route through there... I'm surprised some of that works tbh, I'm sure I tried to manually mount hostapp partitions and even with that I couldn't get them accessible, but I'll do some further investigation.

Either way though, it'll definitely be much much easier if we can expose this in the boot partition like everything else :+1:.

hedss, 2018-02-05:
Ah yes, I'd forgotten about that. Originally it used the `node-lkl`, but that flat out doesn't compile under OSX or Windows. We'd discussed keys being in the state partition in future. This would make a strong argument for doing that.

agherzan, 2018-02-05:
@shaunmulligan if that approach is decided, can you add an issue in meta-resin and assign it to me?

shaunmulligan, 2018-02-05:
@agherzan I think it is the approach we will go with, there is already an issue there from the Hass.io guys: https://github.com/resin-os/meta-resin/issues/608

agherzan, 2018-02-05:
In that case can we close this issue @shaunmulligan ? It seems like we have the one in cli and one in meta-resin.

shaunmulligan, 2018-02-05:
@agherzan yup, thats okay by me. The CLI issue now links to both issues

closed 2018-02-05 by agherzan