balena-os/meta-balena #509: Hostname setting on Cybertan Intel Quark

imrehg, 2016-12-06:
In the logs I see `Set hostname to <021bb9f-021bb9f>.`, which is twice the short UUID, instead of the `devicetype-uuid` pattern as it is for other boards.

imrehg, 2016-12-06:
Looks like this happens on the other devices as well (seen the same issue on the ARTIK 710)

michal-mazurek, 2016-12-13:
I tested 2.0.0-beta.7 on bbb and rpi3, and provided "In the logs" means "journalctl" I cannot reproduce this behaviour.

michal-mazurek, 2016-12-13:
I can recreate this on 2.0.0-beta.7 on a rpi3 and bbb, provided the device is running an app.

agherzan, 2016-12-13:
Ok @michal-mazurek . Given that you can reproduce this, could you please debug and see what's going on? I'm assigning you here.

agherzan, 2016-12-16:
This needs to be handled in the docker images, the issue was created in base-images so we can close this meta-resin one.

closed 2016-12-16 by agherzan