balena-os/meta-balena #514: run-resinhup.sh still points to the old location of supervisor.conf

pcarranzav, 2016-12-07:
See https://github.com/resin-os/meta-resin/blob/master/meta-resin-common/recipes-support/resinhup/resinhup/run-resinhup.sh#L305
Not sure what the consequences might be, but it looks like it could affect architecture detection and therefore the supervisor update and docker images migration could go wrong.

(Also not sure if since this does OS updates from older versions it should actually look in the old *and* new locations? i.e. /etc/supervisor.conf and /etc/resin-supervisor.conf)

floion, 2017-05-24:
ping @agherzan @willnewton

agherzan, 2017-05-24:
The move was only available in v2.0-beta.2 which doesn't support hostOS updates for now. Closing this as it will be handled with implementation of resinhup for 2.0.

closed 2017-05-24 by agherzan