balena-os/meta-balena #519: Current bootloader doesn't support RPi2 v1.2

imrehg, 2016-12-09:
Raspberry Pi 2 has a new hardware version, v1.2, which seems to needs an update of the bootloader in the resin OS image to boot. Tested with the latest bootloader from Raspbian this to be the case.

[forum](https://talk.resin.io/t/rpi-2-is-not-booting-image/431/)

agherzan, 2016-12-09:
@floion Can we update the bootloader (only) in our resin-raspberrypi, and release 1.25 on these boards on Monday? Would you be able to take this task, please? If that's the only change in codebase we don't even need to test the entire suite. Checking that all rpi boards still boot is enough.

imrehg, 2016-12-09:
fyi I'm happy help with the test runs (RPi, RPi2 v1.1, RPi2 v1.2, RPi3)

imrehg, 2016-12-14:
Terminology correction: out of date firmware. Tracked down to updating even just `start_cd.elf` to the latest one shipped in Raspbian image's boot partition made the v1.2 boot.

agherzan, 2016-12-21:
@floion @imrehg please link the resolving PR and close this issue.

closed 2016-12-27 by michal-mazurek