balena-os/meta-balena #548: connman service has requires instead of wants for resin-net-config

lifeeth, 2017-01-20:
Connman service has `requires` instead of `wants` for the resin-net-config service resulting it to stop when resin-net-config stops. resin-net-config is a oneshot service and should not impact connman after it runs once.

floion, 2017-05-24:
I believe this is for the 1.X branch

lifeeth, 2017-05-24:
Yes

agherzan, 2017-05-24:
Why would the resin-net-config stop?

willnewton, 2017-07-17:
This doesn't seem to be the case, and indeed it seems like it always has been "wants" going through git history:

    $ cat meta-resin-common/recipes-connectivity/connman/resin-connman-conf/connman.conf
    [Unit]
    Wants=resin-net-config.service
    After=resin-net-config.service
    
    [Service]
    OOMScoreAdjust=-1000
    ExecStart=
    ExecStart=/usr/sbin/connmand -n --nodnsproxy

I'm not really sure what the intended resolution is but it is probably "upgrade to 2.x".

closed 2017-07-17 by willnewton