balena-os/meta-balena #565: resin.io and resinos has different dbus path

lfarkas, 2017-02-06:
I don't know which part of the system responsible for this but in resin.io the host's dbus system is mounted under /host_run/dbus/system_bus_socket (as described in https://docs.resin.io/runtime/runtime/#dbus-communication-with-hostos) while in resinos /host/run/dbus/system_bus_socket. i hope you see the difference:-) with this difference all program which use dbus and working differently on resin.io and resinos.

floion, 2017-03-03:
Hi,
the next version of resinio will have the host's dbus socket as that in resinos.
Sorry for the inconvenience

agherzan, 2017-03-23:
@floion can you link the PR which made this change and close this issue, please?

pcarranzav, 2017-04-11:
@lfarkas in the newer versions of resin.io (supervisor above 1.7.0), the dbus socket is at `/host/run`. The socket at `/host_run` was deprecated and removed in supervisor 4.0.0. You can use the `/host/run` path both in unmanaged resinOS and resin.io.

closed 2017-04-11 by pcarranzav