balena-os/meta-balena #581: Corrupted image on 2.0.0-beta10.rev1

Page-, 2017-02-22:
A user had a downloaded image that appeared to be corrupted, as it was missing basic files like `/bin/sh` and `/bin/bash`

agherzan, 2017-02-23:
@Page- What board was this encountered on? And did we manage to reproduce ourselves too?

Page-, 2017-02-23:
It was an artik710

agherzan, 2017-02-23:
I just downloaded an image mounted it on top of a loop device:
andrei@carbonA tmp $ losetup 
```
NAME       SIZELIMIT OFFSET AUTOCLEAR RO BACK-FILE                                                                  DIO
/dev/loop1         0      0         1  0 /var/lib/docker/devicemapper/devicemapper/metadata                           0
/dev/loop2         0      0         0  0 /tmp/resinstaging-StagingArtik710-2.0.0-beta11.rev1-3.0.0-e86460b1cc71.img   0
/dev/loop0         0      0         1  0 /var/lib/docker/devicemapper/devicemapper/data                               0
```

After mounting the second partition I see both bash and sh available in the rootfs:

```
andrei@carbonA test $ sudo find . -name bash
./bin/bash
andrei@carbonA test $ sudo find . -name sh
./bin/sh
```

Page-, 2017-02-23:
Sorry, the image I was talking about was a docker image, not a hostos image, I've updated the issue to reflect that

agherzan, 2017-02-24:
What is the docker image you are referring to?

Page-, 2017-02-25:
The user application that gets downloaded, it was fine in the registry and on another device, but it was corrupted on a specific device for some reason - not something I've seen happen on the 1.x devices

shaunmulligan, 2017-02-25:
I think we have seen this on 1.x on the beaglebone black, the user reported certain files randomly missing from their container

agherzan, 2017-02-26:
The report is very vague. We would need to try to reproduce first or bring more details to the reported issue. I've personally never seen this symptom.

petrosagg, 2017-04-18:
This will be fixed by resin-os/meta-resin#643

willnewton, 2017-07-18:
The referenced PR has been merged.

closed 2017-07-18 by willnewton