balena-os/meta-balena #601: run-resinhup.sh: typos

imrehg, 2017-03-16:
There are some typos in the commands and help text of the resinhup helper script

resin-jenkins, 2017-03-16:
Hi there, just a heads up – we’ll be moving support for free users to the [community forums](https://forums.resin.io/c/troubleshooting) 

this Monday, March 20. The main difference is that your questions and our developers' answers will be public instead of private, building a knowledge base for everyone.

You can take this question to the forums now, I’ll meet you there to answer it. If not, I’ll help you out here, but know that future questions should be directed to the forums. 

P.S. if you’d like to continue accessing private support after the change, we can upgrade you to a paid plan and you won’t see any difference.

willnewton, 2017-04-18:
The related pull request has been merged.

closed 2017-04-18 by willnewton