balena-os/meta-balena #608: Sync authorized_keys from boot / Key support for ResinOS only things

pvizeli, 2017-03-24:
Hi 

I create a script on my yocto resin layer, they sync the `autorized_keys` file from boot with `/home/root/.ssh`. I add also a mount overlay script for that folder. For user they use only resinos is the static public key not very usefully.

1) https://github.com/pvizeli/hassio/tree/master/meta-hassio/recipes-core/dropbear
2) https://github.com/pvizeli/hassio/tree/master/meta-hassio/recipes-support/resin-mounts

I think that will be a nice feature for resinos. I think also it Need a new config variable like RESINIO_CONNECTABLE. If that will set it use your public key and is that not set it use the sync script.

agherzan, 2017-04-05:
@shaunmulligan , @petrosagg what do you guys think about this? It could be something we want for development images?

shaunmulligan, 2017-04-05:
@agherzan yeah its definitely something we would want. We need to think about how we would do the authentication of the docker socket, but that is a separate topic.

pvizeli, 2017-04-06:
I'm not sure if a authentication on docker socket very usefully since it is only internal and have a attacker access to this socket it have also access to hole system and don't need this socket. It would be only make some things complexer for 3party stuff like hassio.

pvizeli, 2017-05-18:
- > https://github.com/home-assistant/hassio-build/tree/master/meta-hassio/recipes-support/resin-mounts

- > https://github.com/home-assistant/hassio-build/tree/master/meta-hassio/recipes-core/dropbear

trinitronx, 2017-06-03:
Yeah, general production practice with docker hosts (at least with Kubernetes clusters) is to leave docker on the local unix socket only.  Then system-level services like `kubelet` can access docker over the socket to manage containers.  Security is handled at the layer of `kube-apiserver`, which is a higher-level control plane service that runs on a single node in the cluster.  `kubelet` on each node authenticates to `kube-apiserver` with TLS Client certificates signed by a CA which `kube-apiserver` trusts.  No docker daemon authentication is done.

What @pvizeli says about needing command line access to get to `docker.sock` is true, and relies on local unix file permissions on the socket file.  Usually it is only root who can access, but in practice some like to add a `docker` group which has group `rw` permission on the socket file.

trinitronx, 2017-06-03:
As a user of ResinOS, I just wanted to chime in here to say that making `~/.ssh/` a part of the `resin-overlay` mount is a great feature for securing hardware running ResinOS.  I've been waiting for something like this 😄

shaunmulligan, 2017-06-05:
@agherzan I think this would be a good feature to add. @pvizeli would you consider PR'ing it to meta-resin?

pvizeli, 2017-06-06:
@shaunmulligan I can do that. You want it for dev image only?
As maintainer of hass.io it would be nice if we can produce a productive unmanaged image that have this feature. The actual version of `barys` make it very hard to make image like hass.io.

shaunmulligan, 2017-06-08:
@pvizeli yeah, I think for the `-dev` versions only. the production images are basically identical to the `-dev` images but with the local ssh and docker socket shutdown.

agherzan, 2018-02-02:
@pvizeli @shaunmulligan Do we have any updates on this?

commit 767e1139 (2018-10-17): resin-vars: Implement custom ssh keys service

This patch adds support for translating SSH public keys from config.json
to an authorize_keys file. We use `authorize_keys_local` file as the
destination file of this translation so we can differentiate in between keys
brought locally from config.json and keys brought from the balena backend
through os-config an avoid racing issues with this tool (os-config). The final
`authorize_keys` file will be merged at every ssh connection in the dropbear
service.

These SSH custom public keys in config.json are assumed to be provided
as an array in `.os.ssh [... cut]
commit 7718677d (2018-10-17): resin-vars: Implement custom ssh keys service

This patch adds support for translating SSH public keys from config.json
to an authorize_keys file. We use `authorize_keys_local` file as the
destination file of this translation so we can differentiate in between keys
brought locally from config.json and keys brought from the balena backend
through os-config an avoid racing issues with this tool (os-config). The final
`authorize_keys` file will be merged at every ssh connection in the dropbear
service.

These SSH custom public keys in config.json are assumed to be provided
as an array in `.os.ssh [... cut]
commit 6d2e43d4 (2018-10-17): resin-vars: Implement custom ssh keys service

This patch adds support for translating SSH public keys from config.json
to an authorize_keys file. We use `authorize_keys_local` file as the
destination file of this translation so we can differentiate in between keys
brought locally from config.json and keys brought from the balena backend
through os-config an avoid racing issues with this tool (os-config). The final
`authorize_keys` file will be merged at every ssh connection in the dropbear
service.

These SSH custom public keys in config.json are assumed to be provided
as an array in `.os.ssh [... cut]
closed 2018-10-18 by agherzan