balena-os/meta-balena #639: Improvements of the "Gettings started" guide

mh182, 2017-04-09:
I tried to setup my first ResinOS installation on an Ordoid-C1. Since I have no easy means to install nodejs 6.x I could not use rdt and thus went for the manual configuration. I got stuck/confused in section https://resinos.io/docs/odroid-c1/gettingstarted/#configure-the-image. 

> And edit the ssid and psk values in the /boot/system-connections/resin-sample connection file

Since I have no WIFI but use an ethernet connection, I was not sure if I would get a working network connection in case I do not change the configuration shown in the manual. So I mounted the image on the loopback device and searched for alternative configuration examples for an ethernet card (with no luck).

Two sections below in "Boot the device" I got the answer about my WIFI/ethernet problem:
> You will also need to ensure that the device can connect to the network, either via ethernet or wifi.

Yeah, yeah ... maybe i should have read on before trying to configure the ethernet settings manually. I propose to change "Configure the image" of manual configuration:  point out that an ethernet connection is configured by default and no settings have to be changed in case an ethernet connection is used.

Here the 2nd point:  a link or a description how to change the files manually would be fine. Creating loopback devices, mount it ...

mh182, 2017-04-09:
Some minor improvements:

Section https://resinos.io/docs/odroid-c1/gettingstarted/#install-resin-device-toolbox:
* `node.js 6.x` is listed as dependency - but `node.js 4.8` works fine. The `npm` Package description of `resin-device-toolbox` also shows `NodeJS >= 4`.

Section https://resinos.io/docs/odroid-c1/gettingstarted/#get-a-container-running add 
> $ cd resinos-sample
> $ rdt push resin.local -s .

Just `rdt push resin -s .` did not work. I had to add `.local` to the hostname to resolve it.

floion, 2017-05-08:
@agherzan ^

agherzan, 2017-05-08:
@mh182 sorry for the late reply. @shaunmulligan this sounds like a documentation problem / rdt. Can you confirm? Who should take a look on this?

shaunmulligan, 2017-05-09:
@agherzan these are indeed documentation issues, some of them have already been addressed in updating the docs to use `resin cli` instead of `rdt`, but the ones haven't yet. I'm not sure who can pick this up at the moment, I might have some time later this week.

floion, 2017-05-24:
ping @shaunmulligan

floion, 2017-05-26:
maybe also ping @zwalchuk

zwalchuk, 2017-05-26:
I'll take ownership of this

zwalchuk, 2017-07-18:
Moving to resinos repo, https://github.com/resin-os/resinos/issues/354. Will address with other documentation changes

closed 2017-07-18 by zwalchuk