balena-os/meta-balena #671: NetworkManager service not visible inside the user container

josephroberts, 2017-05-15:
Device type: Intel NUC
OS: Resin OS 2.0.0+rev3 (prod)
Super: v4.1.1

To reproduce:
* Provision an Intel NUC
* Push a user container and wait for it to deploy
* SSH into the user container and run the commands below:
```
export DBUS_SYSTEM_BUS_ADDRESS=unix:path=/host/run/dbus/system_bus_socket
systemctl list-unit-files --type=service
```
* `NetworkManager.service` will not be listed and it should (running the same command in the host returns `NetworkManager.service` as expected.

floion, 2017-05-30:
@josephroberts is this NUC specific?

josephroberts, 2017-05-30:
@floion I haven't tested with other devices yet but I would of expected more reports from users if this was an issue on more popular boards. 

I plan to do some testing this week to see if resinOS1.X has the same issue on the NUC

floion, 2017-05-30:
Maybe also try on a different board too?

shaunmulligan, 2017-05-30:
It might be interesting to try on the UPboard, since we definitely aren't seeing this on the rpi3

richguernsey, 2017-06-01:
What I've seen is the interface name on the NUC is different from the standard "wlan0" simply changing that in the config.coffee file from:::: iface: process.env.PORTAL_INTERFACE or 'wlan0' to:::: iface: process.env.PORTAL_INTERFACE or 'wlp58s0' worked for me.  Using NetworkManager, Linux Image

josephroberts, 2017-06-01:
I think your right, I had assumed that `systemctl list-unit-files --type=service` would list `NetworkManager` but actually this has never been the case. I think we can probably improve upon the interface selection to make it automatic  which should avoid issues like this in the future.

agherzan, 2018-02-02:
This issue is not reproducible on rpi3. If this is still an issue though on other boards feel free to open an issue in resinos repository or in the board specific on. I'll close this for now.

closed 2018-02-02 by agherzan